foss-for-mips-arc-processors/embarc_osp #3: file folder ide_projects missing in your code packet

ZuoJiawei, 2017-06-30:
# Issue Summary
- **Type**: *Enhancement*
- **Category**: *ARC HAL* | *Device HAL* | *BSP* | *OS* | *Middleware* | *Library* | *Example* | *Build System* | *Tools* | *Documentation* | *Installation* | *Usability*
- **Priority**: *Medium* 
- **Release Version**: 2017.06
--------------------------------------------------------------------------------
# Question
   After reading [Embarc Documentation](http://embarc.org/embarc_osp/doc/embARC_Document/html/index.html#sect_intro_overview)
I find an file folder being taged as **ide_projects** listed in the document overview is missing in your latest code release.Did you change this folder's name (if so what is folder's name)or just delete it?

fanghuaqi, 2017-06-30:
Hi Jiawei,

For *ide_projects* folder, it only existed in release package of embARC OSP, you can get the release package [here](https://github.com/foss-for-synopsys-dwc-arc-processors/embarc_osp/releases). If you just clone the repo source code, you can only try in command line mode. Or you can copy the *ide_projects* folder in release package, and pasted in your source code root of embARC OSP.

Thanks
Huaqi

fanghuaqi, 2017-07-03:
Maybe we can take this explanation as FAQs.

closed 2017-07-05 by fanghuaqi