foss-for-mips-arc-processors/embarc_osp #10: Secureshield examples throw exception when build with OLEVEL=O0

jsun85, 2017-07-12:
# Issue Summary
- **Type**: *Bug*
- **Category**: *Library*
- **Priority**: *High*
- **Release Version**: 2017.03

--------------------------------------------------------------------------------
# Bug

## Development Environment
- **HOST OS**
    Ubuntu 16.04 LTS 64 bit

- **TOOLCHAIN**
    ARC GNU 2017.03.

- **BOARD**
    EMSK 2.3

- **ARC CORE**
    EM7D.

## Bug Description
   Take secret_secure under example/baremetal/secureshield as example, other examples share the same issue.
   When build the example using command:
make BOARD=emsk BD_VER=23 CUR_CORE=arcem7d TOOLCHAIN=gnu OLEVEL=O0

In console, any command invoke main cause a exception:

COM1>main -h
soft interrupt 1 interrupt handler
../../../../library/secureshield/core/src/v2/secureshield_sjli.c: Line 121: halt reason:interface @253c0 not found in container 1

But build without OLEVEL=O0 works fine.

The problem is caused by sjli instruction: when invoke sjli instruction, the registers for parameters are set before a call to arc_lock_save (in secureshield_secure_call_exports.h line 103-118), when build with O0, the call to function arc_lock_save is a function call and destroyed r0 and r2, that are the parameters for sjli. When build is O2, gcc optimize the function call out and only generated a clri instruction like this:
     81a:       272f 013f               clri    r4

--------------------------------------------------------------------------------
# Enhancement
   To make the code independent of optimization level, either move the function call before set of parameter registers or use clri and seti instruction directly.

--------------------------------------------------------------------------------
# Question

jsun85, 2017-07-12:
I tried move the call forward and it doesn't work, then I try to directly generate instruction as compiler do, like this:

`#define QUOTE(str) #str`
`#define EXPAND_AND_QUOTE(str) QUOTE(str)`
`#define SECURESHIELD_SECURE_CALL(id, metadata, ...) \`
`	({ \`
`		SECURESHIELD_MACRO_REGS_ARGS(uint32_t, ##__VA_ARGS__) \`
`		SECURESHIELD_MACRO_REGS_RETVAL(uint32_t, res) \`
`		/* uint32_t status = arc_lock_save(); */ \`
`		Asm( \`
`			"clri r" EXPAND_AND_QUOTE(SECURESHIELD_MACRO_NARGS(__VA_ARGS__)) "\n" \`
`			"push_s %%blink\n" \`
`			".long %[secure_call_id]\n" \`
`			metadata \`
`			"pop_s %%blink\n" \`
`			"seti r" EXPAND_AND_QUOTE(SECURESHIELD_MACRO_NARGS(__VA_ARGS__)) "\n" \`
`			: SECURESHIELD_MACRO_ASM_OUTPUT(res) \`
`			: SECURESHIELD_MACRO_ASM_INPUT(__VA_ARGS__), \`
`			[secure_call_id] "i" (SJLI_OPCODE(id)) \`
`			: "r" EXPAND_AND_QUOTE(SECURESHIELD_MACRO_NARGS(__VA_ARGS__)) "" \`
`		); \`
`		/* arc_unlock_restore(status); */ \`
`		res; \`
`	})`
``
`#else`
`#error "unsupported secureshield version"`
`#endif`

This approach works, so I post it here for your reference.

vonhust, 2017-07-17:
already fixed

closed 2017-07-17 by vonhust