foss-for-mips-arc-processors/embarc_osp #11: A question about Bootloader

XiangcaiHuang, 2017-07-19:
# Issue Summary
- **Type**: *Question*
- **Category**: *Example*
- **Priority**:*Low*
- **Release Version**: 2017.03

## Development Environment
- **HOST OS**
    Windows 7, 64Bit

- **TOOLCHAIN**
    ARC GNU 2017.03

- **BOARD**
    EMSK 2.2

- **ARC CORE**
    EM11D

# Question
   In the user manual of the example, embARC Secondary Bootloader, it says that "For EMSK1.x, bootloader core configuration must be arcem6, for EMSK2.x, bootloader core configuration must be arcem7d. ", But why? It was running successfully when I used the core configuration arcem11d for EMSK2.2.

XiangcaiHuang, 2017-07-19:
How the Secondary Bootloader program works? I have take some time to learn it, describing as follows:
1. Primary Bootloader running on the ICCM, it loads the Secondary Bootloader program in the SPI Flash into DDR.
2. Running Secondary Bootloader program on the DDR, it loads the boot.bin or boot.hex in the SD Card into DDR.
3. Running the user program on DDR, There are 2 programs on the DDR at the same time, but they do not affect each other.

fanghuaqi, 2017-07-21:
- For running successfully when you used the core configuration for EMSK 2.2 EM11D, maybe you build the bootloader for EM11D, and put it in SPIFlash, or the EM11D core configuration is not quite different with EM7D core configuration, so the program can run on EM7D, also can run on EM11D for some use cases. And all the program is built optimized for different core configuration.
- For the 3rd question, the 2 programs on the DDR are put in different places, so it won't affect each other's function.

fanghuaqi, 2017-07-24:
Fixed

closed 2017-07-24 by fanghuaqi