foss-for-mips-arc-processors/embarc_osp #13: Bug about dma_spiflash example

XiangcaiHuang, 2017-07-20:
# Issue Summary
- **Type**: *Bug*
- **Category**:*Example*
- **Release Version**: 2017.03

--------------------------------------------------------------------------------
# Bug

## Development Environment
- **HOST OS**
    Windows 7, 64Bit

- **TOOLCHAIN**
    ARC GNU 2017.03.

- **BOARD**
    EMSK 2.2

- **ARC CORE**
    EM7D.

## Bug Description
   I want to run baremetal_dma_spiflash example but it doesn't work, it just output the working frequency &1000000 and random vlaue 4, then got struck. What went wrong?

fanghuaqi, 2017-07-20:
This only works with EMSK 2.3, which DW SPI has uDMA interface.

fanghuaqi, 2017-07-21:
Add notice about support status in this example's document extra comment later.

fanghuaqi, 2017-07-24:
Fixed

closed 2017-07-24 by fanghuaqi