foss-for-mips-arc-processors/embarc_osp #15: Wrong data type in secureshield.h

jsun85, 2017-07-23:
# Issue Summary
- **Type**: *Bug*
- **Category**: *Library*
- **Priority**: *High*
- **Release Version**: 2017.03

--------------------------------------------------------------------------------
# Bug

## Development Environment
- **HOST OS**
    Ubuntu 16.04 LTS 64Bit

- **TOOLCHAIN**
    ARC GNU 2017.03.

- **BOARD**
    EMSK 2.3

- **ARC CORE**
    EM7D.

## Bug Description
   In secureshield.h line 65 and 66:

	uint32_t *background_rom_start, background_rom_end;
	uint32_t *background_ram_start, background_ram_end;

A asterisk symbol should be put before background_rom/ram_end to make them pointer.

--------------------------------------------------------------------------------
# Enhancement

--------------------------------------------------------------------------------
# Question

jsun85, 2017-07-23:
Also a typo in secureshield_vmpu_exports.h Line 126

#define SECURESHILED_CONTAINER_NORMAL	0x0 			/* container type: normal */

SECURESHILED should be SECURESHIELD.

vonhust, 2017-07-24:
fixed in commit ef63996

jsun85, 2017-07-25:
Commit ef63996 fixed the typo I wrote in the comment, but did not fix the wrong data type in secureshield.h. You can see without a * in the front of background_rom_end and background_ram_end, they are declared as 32 bit integers but not a pointer of 32 bit integer.

fanghuaqi, 2017-07-26:
@jsun85 You can also send pull request of your changes, @vonhust can review it and see if can be merged into our development. Thanks for submitting issues to embARC OSP.

fanghuaqi, 2017-08-01:
Fixed by adding missing asterisk symbol to `background_rom_end` and `background_ram_end`, thanks @jsun85  for your contribution.

closed 2017-08-01 by fanghuaqi