foss-for-mips-arc-processors/embarc_osp #17: secret_secure_sid example raise protection violation with stack operations

jsun85, 2017-07-27:
# Issue Summary
- **Type**: *Bug*
- **Category**:  *Example*
- **Priority**: *Medium*
- **Release Version**: Latest commit b8d2f7de9c091cdeddf95054f9fa6eddd47b81cb

--------------------------------------------------------------------------------
# Bug

## Development Environment
- **HOST OS**
    Ubuntu 16.04 LTS, 64Bit

- **TOOLCHAIN**
    ARC GNU 2017.03.

- **BOARD**
    EMSK 2.3

- **ARC CORE**
    EM7D.

## Bug Description
   Take example secret_secure_sid, build with OLEVEL=O0, that will generate couple of push and pop instructions inside function trusted_ops in container2.c, then run the example with command: main -t, you can see something like this:

COM1>main -t
soft interrupt 1 interrupt handler
soft interrupt 1 interrupt handler
Requesting operation from trusted container
protection violation exception at 0x800
 cause:memory write violation, parameter:0x44
the region for 0x80005714 is not found
../../../../library/secureshield/core/src/v2/secureshield_int.c: Line 224: halt reason:halt caused by protection violation

0x800 is the address of a push blink instruction, and 0x80005714 is the SP value when trusted_ops is called. It is within the region of _f_bss_container2 and _e_bss_container2. So I think set of this 2 values does not take effect. But when build with O2,

		ctx->initialized = 1;

is fine, that line actually write to background_container's bss region.

When build with O2, all the stack operation's is optimized away, so the example works fine. To reproduce this error with O2, you can add:

	Asm("push_s blink\n");
	Asm("pop_s blink\n");

inside trusted_ops.

--------------------------------------------------------------------------------
# Enhancement

--------------------------------------------------------------------------------
# Question

jsun85, 2017-07-30:
I found the problem is the secureshield configuration for this example does not generate AC table entry for text and rom data section for container2, but only generate a entry for data and bss section:

container[2] AC table:
        container[2] ac[7]={0x80005340,size=1024,ac=0x80000136}
        container[2] has interface @0x824 with 0 arguments
 
Thanks to your test_case example which generate proper AC table for all containers. I modified secureshield_appl_config.h accordingly, here is the diff:

diff --git a/example/baremetal/secureshield/secret_secure_sid/secureshield_appl_config.h b/example/baremetal/secureshield/secret_secure_sid/secureshield_appl_config.h
index cec1c67..58f2436 100644
--- a/example/baremetal/secureshield/secret_secure_sid/secureshield_appl_config.h
+++ b/example/baremetal/secureshield/secret_secure_sid/secureshield_appl_config.h
@@ -62,9 +62,20 @@
/* tell linker script template, there is a region named container1 in secure rom, its size is 0 */
#define SECURESHIELD_REGION_SECURE_CONTAINERS_ROM GEN_SECURE_CONTAINER_ROM_SECTION(container1, 0)

+// CONTAINER_ROM_OBJS_LIST is only for ARC_GNU
+// This macro wii exclude the specified container objs from background container in linking,
+// *(EXCLUDE_FILE(CONTAINER_ROM_OBJS_LIST) .text EXCLUDE_FILE(CONTAINER_ROM_OBJS_LIST) 
+// .text.* EXCLUDE_FILE(CONTAINER_ROM_OBJS_LIST) .gnu.linkonce.t.*)
+#define CONTAINER_ROM_OBJS_LIST *container2.o

/* tell linker script template, there is a region named container1 in secure ram, its size is 2048 bytes */
#define SECURESHIELD_REGION_SECURE_CONTAINERS_RAM GEN_SECURE_CONTAINER_RAM_SECTION(container1, 2048)
 
+// CONTAINER_RAM_OBJS_LIST is only for ARC_GNU
+// This macro wii exclude the specified container objs from background container in linking,
+//*(EXCLUDE_FILE(CONTAINER_RAM_OBJS_LIST) .data EXCLUDE_FILE(CONTAINER_RAM_OBJS_LIST) 
+//.data.* EXCLUDE_FILE(CONTAINER_RAM_OBJS_LIST) .gnu.linkonce.d.*)
+#define CONTAINER_RAM_OBJS_LIST *container2.o
 
/* tell linker script template, there is a region named container2 in normal rom, its size is 0 bytes */
#define SECURESHIELD_REGION_CONTAINERS_ROM GEN_CONTAINER_ROM_SECTION(container2, 2048, *container2.o*)

Note: looks like the content within double slash is use by linker script.
Now everything works fine:

container[2] AC table:
        container[2] ac[7]={0x1bbc0,size=96,ac=0x12d}
        container[2] ac[8]={0x1bc20,size=32,ac=0x124}
        container[2] ac[9]={0x80005340,size=1024,ac=0x136}
        container[2] has interface @0x1bbe4 with 0 arguments

fanghuaqi, 2017-08-01:
Hi @jsun85 , I just post a commit using your patch provided, tested OK on both toolchains, thanks for your contribution.

closed 2017-08-01 by fanghuaqi