foss-for-mips-arc-processors/embarc_osp #21: secureshield v1 bug

Jeff006, 2017-11-06:
# Issue Summary
- **Type**: *Bug*
- **Category**: *Library**Usability*
- **Priority**:  *High*
- **Release Version**: 2017.03

--------------------------------------------------------------------------------
# Bug

## Development Environment
- **HOST OS**
All

- **TOOLCHAIN**
All

- **BOARD**
All

- **ARC CORE**
All

## Bug Description
In secureshiled.h, struct SECURESHIELD_CONFIG defines mode as a pointer to a 32 bits word:
typedef struct {
	uint32_t magic;
	uint32_t *mode;

	/* protected flash memory region */
	uint32_t *secure_rom_start, *secure_rom_end;
	/* protected bss */
	uint32_t *secure_ram_start, *secure_ram_end;

	uint32_t *background_rom_start, background_rom_end;
	uint32_t *background_ram_start, background_ram_end;

	/* container configuration tables */
	uint32_t *cfgtbl_start, *cfgtbl_end;

	/* pointers to container configuration tables */
	uint32_t *cfgtbl_ptr_start, *cfgtbl_ptr_end;
} EMBARC_PACKED SECURESHIELD_CONFIG;

Unfortunately, in wrapper.s where the structure is actually initialized, a value is written (0 or 1) instead of an address of a 32 bits word. Any dereferencing of the pointer "mode" can lead to a memory access failure (bad memory address or mpu protected address). In our case, in secureshield.c, the following line:
*(__secureshield_config.mode) = SECURESHIELD_DISABLED;
tries to write in ROM at address 0x1 (flash memory) which is not allowed and raises an exception.

--------------------------------------------------------------------------------
# Enhancement
Two options are possible:
-Declares "mode" as a 32 bits word but as it resides in secure ROM, it will be constant.
-Actually initialize "mode" with the address of a 32 bits word in secure RAM, the later taking the value 0 or 1. I don't know in this case if it is usefull to store the address in ROM inside SECURESHIELD_CONFIG or better to directly use the variable in secure RAM...

closed 2017-11-06 by Jeff006

Jeff006, 2017-11-06:
I was wrong, sorry. An address is actually written in wrapper.s, in our case the address is 0x10000000 and with byte swapping I mixed it up with a the value "1". Though we have some issue with GCC, in wrapper.s when linker resolves the addresses in secureshield_config, the swapping is bad. It swaps bytes on 16 bits words...

fbedard, 2017-11-06:
If you need assistance with your GNU linker problem, you can file a separate issue for it at 
https://github.com/foss-for-synopsys-dwc-arc-processors/toolchain/issues 

Thanks.