foss-for-mips-arc-processors/embarc_osp #24: IoT DK UART1 and UART2 cannot be used in interrupt mode

YuguoWH, 2018-02-28:
# Issue Summary
- **Type**: *Bug*
- **Category**: *Device HAL*
- **Priority**: *Medium*
- **Version**:  94ee519  at branch feature/iotdevkit

--------------------------------------------------------------------------------
# Bug

## Development Environment
- **HOST OS**
   Windows 10, 64Bit

- **TOOLCHAIN**
 MWDT build (2017.09/rel)

- **BOARD**
    IOTDK

- **ARC CORE**
    arcem9d

## Bug Description
   The TX of UART1 and UART2 are fine but will hang trying to receive any message with interrupt. Seems like signals are unable to be received in UART1&2.

qiangg, 2018-03-02:
The driver is out-of-date and will be updated soon.

vonhust, 2018-12-12:
already fixed

closed 2018-12-12 by vonhust