foss-for-mips-arc-processors/embarc_osp #27: Some network and ntshell examples are not available for IoT DK

YuguoWH, 2018-02-28:
# Issue Summary
- **Type**: *Bug*
- **Category**: *Example*
- **Priority**: *Medium*
- **Version**:  94ee519 at branch feature/iotdevkit

--------------------------------------------------------------------------------
# Bug

## Development Environment
- **HOST OS**
   Windows 10, 64Bit

- **TOOLCHAIN**
 MWDT build (2017.09/rel)

- **BOARD**
    IOTDK

- **ARC CORE**
    arcem9d

## Bug Description
  I've tried examples provided in example folder, however some network example requires ntshell whereas ntshell example is not support for IOTDK yet.

qiangg, 2018-03-02:
The reason is the memory limitation of IoT DK. Need to optimize the network and shell examples to reduce the size of them.

vonhust, 2018-12-12:
already fixed

closed 2018-12-12 by vonhust