foss-for-mips-arc-processors/embarc_osp #30: Question

willtuna, 2018-03-15:
# Issue Summary
- **Type**:*Enhancement*
- **Category**:  *Example* 
- **Priority**: *Urgent* | *High* | *Medium* | *Low*
- **Release Version**: 2017.12

--------------------------------------------------------------------------------

## Development Environment
- **HOST OS**
    Windows 10

- **TOOLCHAIN**
    ARC GNU 2017.09.

- **BOARD**
    EMSK 1.1

- **ARC CORE**
    EM6

--------------------------------------------------------------------------------
# Question
  Is there any example for GPIO ?
(like how to mapped to the defined pmod pin for reading and some for write) 

Since I have to connect the EMSK with my custom module. So I need to read and write synchronous with clock through the pmod in EMSK with baremetal development.

qiangg, 2018-04-13:
Hi,

Please see the <embARC>/example/blinky . The EMSK LED blinky driver is in <embARC>/board/emsk/common/emsk_gpio.c and .h.

YuguoWH, 2018-04-13:
Hi @willtuna , 
Sorry for late reply. Basically you could use PORT and MASK to choose your GPIO pin or pins, use it as parameter and call gpio_get_dev().

fanghuaqi, 2018-04-14:
See doc links:
* [Device HAL](http://embarc.org/embarc_osp/doc/embARC_Document/html/group___d_e_v_i_c_e___h_a_l___d_e_f.html)
* [GPIO Device HAL](http://embarc.org/embarc_osp/doc/embARC_Document/html/group___d_e_v_i_c_e___h_a_l___g_p_i_o.html)

For example usage, see [EMSK Button LED Switch](http://embarc.org/embarc_osp/doc/embARC_Document/html/emsk__gpio_8c_source.html#l00091)

closed 2018-05-16 by vonhust