foss-for-mips-arc-processors/embarc_osp #34: Question about gpio interrupt and PWM setting

tyhucosiii, 2018-05-22:
# Issue Summary
- **Type**: *Bug* | *Enhancement* | *Question*
- **Category**: *ARC HAL* | *Device HAL* | *BSP* | *OS* | *Middleware* | *Library* | *Example* | *Build System* | *Tools* | *Documentation* | *Installation* | *Usability*
- **Priority**: *Urgent* | *High* | *Medium* | *Low*
- **Release Version**: Specified version such as 2016.05 or All

--------------------------------------------------------------------------------
# Bug

## Development Environment
- **HOST OS**
 Windows 7, 64Bit 

- **TOOLCHAIN**
ARC GNU 2017.09.

- **BOARD**
 EMSK 2.2 

- **ARC CORE**
EM7D.

## Bug Description
I have read the "ibaby" example,and i tried to code the gpio interrupt by following the btn interrupt,(embarc_applications/ibaby_smarthome_multinode/src/lamp_node/driver/btn/) ,but my project could not raise the interrupt.

--------------------------------------------------------------------------------
# Enhancement
```
void usart4_init(void)
{
	DEV_GPIO_BIT_ISR bit_isr;
	DEV_GPIO_INT_CFG int_cfg;

	uart4 = gpio_get_dev(EMSK_GPIOUART4_PORT);

	EMSK_GPIO_CHECK_EXP_NORTN(uart4 != NULL);
	if (uart4->gpio_open(GPIO_UART4_MASK) == E_OPNED) {
		uart4->gpio_control(GPIO_CMD_SET_BIT_DIR_OUTPUT, (void *)(GPIO_UART4OUT_MASK));
		uart4->gpio_control(GPIO_CMD_SET_BIT_DIR_INPUT, (void *)(GPIO_UART4IN_MASK));
		uart4->gpio_control(GPIO_CMD_DIS_BIT_INT, (void *)(GPIO_UART4_MASK));
	}
	int_cfg.int_bit_mask = GPIO_UART4IN_MASK;
	int_cfg.int_bit_type = GPIO_UART4IN_MASK;
	int_cfg.int_bit_polarity = BTN_ACTIVE_LOW;
	int_cfg.int_bit_debounce = GPIO_UART4IN_MASK;
	uart4->gpio_control(GPIO_CMD_SET_BIT_INT_CFG, &int_cfg);
	bit_isr.int_bit_ofs = GPIO_UART4IN_MASK;
	bit_isr.int_bit_handler = EXTI1_IRQHandler;
	uart4->gpio_control(GPIO_CMD_SET_BIT_ISR, &bit_isr);
	uart4->gpio_control(GPIO_CMD_ENA_BIT_INT, (void *)GPIO_UART4IN_MASK);
	uart4->gpio_write(GPIO_UART4OUT_MASK,GPIO_UART4OUT_MASK);
error_exit:
	return;
}

```

--------------------------------------------------------------------------------
# Question
   By the way,I want to konw is there any example about PWM settings?

tyhucosiii, 2018-05-22:
I am using  GPIO PORT A.But my program could not raise the interrupt.
`#define EMSK_GPIOUART4_PORT		DEV_GPIO_PORT_A
#define GPIO_UART4_MASK		0x00030000		//portA的16-17Pin，接J3的7-8口
#define GPIO_UART4_LOW	  (0)
#define GPIO_UART4IN_MASK	0X00010000		//portA的16Pin，接J3的7口
#define GPIO_UART4OUT_MASK	0X00020000		//portA的17Pin，接J3的8口`

fanghuaqi, 2018-05-22:
Please check the default pinmux settings in https://github.com/foss-for-synopsys-dwc-arc-processors/embarc_osp/blob/master/board/emsk/drivers/mux/mux.c#L168, make sure the pins you want to control is set to GPIO not other functions.

And the `bit_isr.int_bit_ofs = GPIO_UART4IN_MASK;` should be offset not mask. Please check and modify it correctly. 

See document about device hal for GPIO here: 
* http://embarc.org/embarc_osp/doc/embARC_Document/html/structdev__gpio__bit__isr.html
* http://embarc.org/embarc_osp/doc/embARC_Document/html/group___d_e_v_i_c_e___h_a_l___g_p_i_o.html

closed 2018-05-23 by fanghuaqi