foss-for-mips-arc-processors/embarc_osp #36: spelling mistake in embARC guide documentation.

IRISZZW, 2018-07-05:
# Issue Summary
- **Type**: *Bug* 
- **Category**:  *Documentation* 
- **Priority**:  *Low*
- **Release Version**: 2017.12

--------------------------------------------------------------------------------
# Bug

## Bug Description
   there is a spelling mistake in embARC guide documentation.
   location is embarc_osp/doc/documents/middleware/middleware.rst.
   line 3 in middleware.rst spelled "Middlwares".
   this spelling mistake could be seen signally  as it is a title of this 
   webpage(http://embarc.org/embarc_osp/doc/build/html/middleware/middleware.html).
[image]

BabaYB, 2018-07-09:
I got  two more spelling mistakes in the doucument.
- ARC processors/Board level start up/OS/`Create WIFI if` seems there is a unnecessary dot
- Libraries/secureshield/6.Compiling&debug/`linker_xx.ldf: the generated linker script for normal applicatio` word `applicatio` missed a `n`

BabaYB, 2018-07-09:
And many HTML labels are used but not works.
For example, in /example/barematel/simple_bootloader, label `<em>` & `<b>` as following
---
[image]
[image]

YuguoWH, 2018-07-09:
Hi @IRISZZW, could you please _fork_ the repo and then use _Pull Request_ to help submit your fixes, thank you very much.

fanghuaqi, 2018-07-09:
@IRISZZW I have merged the pull request #38  that fixed 1 issue, please create a new pull request to fix other issues submit by others.

fanghuaqi, 2018-07-18:
Fixed in pull request #41

closed 2018-07-18 by fanghuaqi