foss-for-mips-arc-processors/embarc_osp #64: Understanding how printing to terminal (UART) works

asampadeh, 2018-10-11:
# Issue Summary
- **Type**: *Question*
- **Category**: *Example* | *Build System* 
- **Priority**: *Medium*
- **Release Version**: Specified version such as 2016.05 or All

--------------------------------------------------------------------------------

# Question
   I want to know how printf() or EMBARC_PRINTF works, and print the output to computer terminal.

asampadeh, 2018-10-11:
I read dw_uart.c or dw_uart_obj.c but still can not relate to printf()

IRISZZW, 2018-10-11:
EMBARC_PRINTF --> xprintf --> xvprintf --> xputc or xputs --> xfunc_out --> console_putchar

[xfunc_out] is an function pointer pointed to [console_putchar] which is initialized in xprintf_setup() called in [board_main] defined in board.c(line 185).

MID_COMMON is defined if MID_SEL = common is defined in your project makefile.

You can follow my lead to review the source code

asampadeh, 2018-10-11:
@IRISZZW  thanks for the link, i will study that.

asampadeh, 2018-10-11:
Now I am reading dw_uart_0_control() in dw_uart_obj.c  ---> calling dw_uart_control(...., ctrl_cmd, ...).
I do not see how dw_uart_install() passing many ctrl_cmd to  dw_uart_0_control() ?

I am not familiar with driver codes :-)

YuguoWH, 2018-10-12:
dw_uart_install() does NOT pass ctrl_cmd, instead it calls dw_uart_0_install() (for example) to init function pointers, see [here in dw_uart_obj.c](https://github.com/foss-for-synopsys-dwc-arc-processors/embarc_osp/blob/master/board/emsk/drivers/ip/designware/uart/dw_uart_obj.c#L128).

asampadeh, 2018-10-12:
Hi @YuguoWH , yes it is calling dw_uart_0_install, but inside the dw_uart_control, many cases based on switch(ctrl_cmd). I cannot see where these ctrl_cmd parameters passed to this API.

YuguoWH, 2018-10-12:
Hi, @asampadeh, whenever a DEV_UART_PTR->uart_control() is called, its calling dw_uart_0_control()/dw_uart_1_control()/dw_uart_2_control(). Please notice that the parameters of DEV_UART_PTR->uart_control() are (uint32_t ctrl_cmd, void *param), which will pass ctrl_cmd to dw_uart_control(). All commands and the prototype of uart_control() are defined in [dev_uart.h](https://github.com/foss-for-synopsys-dwc-arc-processors/embarc_osp/blob/master/device/ip/ip_hal/inc/dev_uart.h)

closed 2019-04-18 by vonhust