foss-for-mips-arc-processors/embarc_osp #65: uDMA memcopy operation  with unaligned address

asampadeh, 2018-10-11:
# Issue Summary
- **Type**:  *Question*
- **Category**: *Example* 
- **Release Version**: Specified version such as 2016.05 or All

--------------------------------------------------------------------------------

# Question
   Anybody knows how to do memcopy using (uDMA controller) with unaligned source or destination address (single transfer mode)? Most important is descriptor setting.
For example src = 0x0000_4001, dst = 0x0000_5000, len = 256 bytes.

If both src and dst addresses aligned (to 4 bytes boundary) then it has no issue.
I am using ArcEm ver 2 core.
Thanks.

YuguoWH, 2018-10-12:
May I know what is the issue and how could I reproduce it? 
BTW I tried the baremetal\arc_feature\udma example with an unaligned address but everything looks fine to me.

asampadeh, 2018-12-07:
Hi @YuguoWH , exactly what I wrote on original thread. Src address is 0x4001 and dst 0x5000. Is memcpy can do the job

0x4001: aa bb cc dd .....
0x4100: dd
after mcpy
0x5000: aa bb cc dd ..
0x50ff: dd

Something like that. Sorry for ignoring this thread since i just open the github again today.

YuguoWH, 2018-12-07:
Hi @asampadeh , last time I did a experiment to copy unaligned address and it worked fine. So I think you just need to provide your address to udma and it will accept the unaligned address. For usage of udma, please refer to `embarc_osp/example/baremetal/arc_feature/udma`

closed 2019-07-26 by vonhust