foss-for-mips-arc-processors/embarc_osp #71: How to start ARC image from DDR beyond 0x0

asampadeh, 2018-12-07:
# Issue Summary
- **Type**:  *Question*
- **Category**: *Build System* 
- **Priority**: *Urgent*
- **Release Version**: metaware_vM-2016.12

--------------------------------------------------------------------------------
# Bug

## Development Environment
- **HOST OS**
    Input the host os name and its version.
    For example, Windows 7, 64Bit | Debian Linux 8.6, 64Bit

- **TOOLCHAIN**
metaware_vM-2016.12

- **BOARD**
nSIM

- **ARC CORE**
EM7D.

## Bug Description

--------------------------------------------------------------------------------
# Enhancement

--------------------------------------------------------------------------------
# Question
I tried simply by changing the LCF file to show DDR region from 0x00010000 instead starting from 0, but only the main.dis and main.hex starting address changed to 00001000 but main.bin (which is going to real DDR chip) still holding the address of 0.
Anybody can help with the compile/linker option to achieave this?

Thanks for advise

vonhust, 2018-12-07:
Do you use embARC_osp ? As I know, embarc_osp will not generate main.dis, also no LCF file.

asampadeh, 2018-12-08:
Hi @vonhust , no, I am using metaware debugger environment

vonhust, 2018-12-10:
@asampadeh  it's a little difficult to reproduce your issuse as we are the maintainer of embarc osp.

vonhust, 2018-12-12:
not an embarc_osp related issue

closed 2018-12-12 by vonhust