foss-for-mips-arc-processors/embarc_osp #92: Clock output using PWM interface

ericwu13, 2019-05-02:
# Issue Summary
- **Type**: *Question*
- **Category**: *Library*
- **Priority**: *High*
- **Release Version**: All

--------------------------------------------------------------------------------
# Question
Is ARC able to output clock using PWM interface? For example, like arduino, we can use PWM output and configure it as 8MHz clock. If yes, is there any library we can use since in embarc website, we can’t find any. Thanks

Below is How Arduino configures a 8MHz clock using AVR library.
```C
DDRB|=(1<<3);//pin 11

ASSR &= ~(_BV(EXCLK) | _BV(AS2));

TCCR2A=(1<<COM2A0)|(1<<WGM21)|(1<<WGM20);

TCCR2B=(1<<WGM22)|(1<<CS20);

OCR2A=0;//(F_CPU)/(2*(X+1))
```

IRISZZW, 2019-05-05:
iotdk board has a 6 channel pwm_timer module, related code in:
board/iotdk/drivers/ip/designware/pwm_timer
device/ip/designware/pwm_timer
device/ip/ip_hal/inc

ericwu13, 2019-05-06:
May I ask about its highest possible PWM clock output rate? Thanks!

IRISZZW, 2019-05-07:
```
int main(void)
{

	io_arduino_config(ARDUINO_PIN_3, ARDUINO_PWM, IO_PINMUX_ENABLE);//pwm timer ch0
	DEV_PWM_TIMER_CFG ch0_pwm_cfg;
	ch0_pwm_cfg.mode = DEV_PWM_TIMER_MODE_PWM;
	ch0_pwm_cfg.count_high = 500;
	ch0_pwm_cfg.count_low = 500;
	ch0_pwm_cfg.isr_hander = NULL;
	DEV_PWM_TIMER_PTR pwm_timer_test = pwm_timer_get_dev(DW_PWM_TIMER_0_ID);
	pwm_timer_test->pwm_timer_open();
	pwm_timer_test->pwm_timer_control(0, PWM_TIMER_CMD_SET_CFG, (void *)(&ch0_pwm_cfg));
	while(1);
}
```

you can do some test. the clock of pwm is 144 MHz
output rate = 144MHz / (count_high + count_low)
duty = count_high / (count_high + count_low)

then number of count_high and count_low should not be to small.

ericwu13, 2019-05-09:
thanks! I can use PWM now. really appreciate your help.

closed 2019-05-09 by ericwu13