foss-for-mips-arc-processors/embarc_osp #96: Maximum console uart baud rate

ericwu13, 2019-05-09:
# Issue Summary
- **Type**: *Question*
- **Category**: *Usability*
- **Priority**: *Urgent* 
- **Release Version**: Specified version such as 2016.05 or All
- **BOARD**  IoTdk
--------------------------------------------------------------------------------
# Question
   the default baudrate is 115200. However, I want a faster baudrate. I set it to 230400 up to 921600. But none of them work. But I can set it to 38400, and my computer serial works. Is it possible to use the console uart baudrate up to 1M? thanks.

PS: I set the baudrate in iotdk.h file like the below.
```C
#define BOARD_CONSOLE_UART_BAUD			UART_BAUDRATE_921600
```

ericwu13, 2019-05-14:
Hi, is there any comments about it? Thanks!

IRISZZW, 2019-05-14:
baud rate = (serial clock frequency)/(16×divisor), and serial clock frequency is 16MHz in itodk board.
so divisor = 10000000/baud rate. when baud rate is high, then divisor will be inaccuracy.

ericwu13, 2019-05-14:
So, is it possible to configure a higher baud rate other than 115200 (e.g. 1M)? If I should change the divisor register in Iotdk, how should I do it? I want a faster serial transmission. Thanks!

jelly99709, 2019-05-16:
I also encounter the baud rate problem. What should I do to change the divisor value in iotdk to have higher baud rate?

ericwu13, 2019-05-27:
Hi, sorry for keeping asking. Is there really no methods to accelerate the baud rate to 1M? I really need a solution for it, but I can't find any... There's little documentation talking about this board. Can you help us on this? Thanks.

IRISZZW, 2019-05-27:
`#define BOARD_CONSOLE_UART_BAUD 1500000 //(iotdk.h)`

```
/** convert uart baudrate to subsystem divisor */
#define SS_UART_BAUD2DIV(perifreq, baud)		((perifreq) / ((baud)*16) + 1)
Inline int32_t set_baud(SS_UART_DEV_CONTEXT *ctx, uint32_t baud)
{
	uint32_t param;
	DEV_UART_INFO *info = ctx->info;

	param = SS_UART_BAUD2DIV(ctx->bus_freq, baud); //param=1 when baud = 1500000
	io_uart_ioctl(ctx->dev_id, IO_UART_SET_DLx, &param);
	info->baudrate = baud;

	return E_OK;
}
```

when you set BOARD_CONSOLE_UART_BAUD  = 1500000, you will get the 1000000 baudrate. 

[image]

ericwu13, 2019-05-27:
why there exists such a difference? Shouldn't we set the baud rate to 1M if we want 1M baud rate? I tried to calculate the divisor value when baud rate = 1.5M and 1M, which are 0.67 and 0.99 respectively. These are different divisor values though. Could you elaborate on how it works? Thanks

IRISZZW, 2019-05-27:
the divisor values should be integer, we round up the divisor values.
`#define SS_UART_BAUD2DIV(perifreq, baud)		((perifreq) / ((baud)*16) + 1)`

ericwu13, 2019-05-27:
oh, so we need divisor = 1, so basically, I can also set baud rate to 1300000 with which we can get divisor value = 1?

IRISZZW, 2019-05-27:
yes, because serial clock frequency is 16MHz, so the max baudrate is 1000000 when divisor = 1.

ericwu13, 2019-05-27:
Thanks. It works!!! I finally understand how to control it. Really appreciate for your help. I'm closing this.

closed 2019-05-27 by ericwu13