foss-for-mips-arc-processors/embarc_osp #97: Interrupt example for Iotdk

ericwu13, 2019-05-09:
# Issue Summary
- **Type**: *Question*
- **Category**: *Example*
- **Priority**: *Urgent*
- **BOARD** Iotdk
--------------------------------------------------------------------------------
# Question
   Could you provide the falling edge interrupt example for iotdk board in arduino pin AD0 and AD1? I can't find any example of this. There is one issue here talking about this, but it seems to be for button interrupt. Thanks.

ericwu13, 2019-05-09:
I'm setting interrupt for arduino AD0 like this
```C
#define PIN_VSYNC 7
int main() {
     io_arduino_config(ARDUINO_PIN_AD0, ARDUINO_GPIO, IO_PINMUX_ENABLE); // vsync
     clk_ptr = gpio_get_dev(DFSS_GPIO_8B3_ID);
     if (clk_ptr->gpio_open((1 << PIN_VSYNC)) == E_OPNED) {
	     clk_ptr->gpio_control(GPIO_CMD_DIS_BIT_INT,
	     (void *)(1 << PIN_VSYNC));
     }
     DEV_GPIO_BIT_ISR bit_isr_vsync;
     DEV_GPIO_INT_CFG int_cfg_vsync;
     int_cfg_vsync.int_bit_mask = 1 << PIN_VSYNC;
     int_cfg_vsync.int_bit_type = GPIO_INT_EDGE_TRIG;
     int_cfg_vsync.int_bit_polarity = GPIO_INT_FALLING_EDGE;
     int_cfg_vsync.int_bit_debounce = GPIO_INT_NO_DEBOUNCE;
     clk_ptr->gpio_control(GPIO_CMD_SET_BIT_INT_CFG, &int_cfg_vsync);

     bit_isr_vsync.int_bit_ofs = PIN_VSYNC;
     bit_isr_vsync.int_bit_handler = vsyncHandler;

     clk_ptr->gpio_control(GPIO_CMD_SET_BIT_ISR, &bit_isr_vsync);
     clk_ptr->gpio_control(GPIO_CMD_ENA_BIT_INT, (void *)(1 << PIN_VSYNC));
}
```

ericwu13, 2019-05-16:
Is there any comments? I wonder if my int_bit_ofs is set correctly. Is this represent the PIN number of the interrupt PIN?

IRISZZW, 2019-05-16:
you can reference the GPIO related code in pmwifi.c.

```
int main()
{
	io_arduino_config(ARDUINO_PIN_AD0, ARDUINO_GPIO, IO_PINMUX_ENABLE); // vsync
	clk_ptr = gpio_get_dev(PIN_PORT);
	if (clk_ptr->gpio_open((1 << PIN_VSYNC)) == E_OK) {
		clk_ptr->gpio_control(GPIO_CMD_SET_BIT_DIR_INPUT, (void *)(1 << PIN_VSYNC));
	}
	int_cfg_vsync.int_bit_mask = 1 << PIN_VSYNC;
	int_cfg_vsync.int_bit_type = GPIO_INT_BIT_LEVEL_TRIG(PIN_VSYNC);
	int_cfg_vsync.int_bit_polarity = GPIO_INT_BIT_POL_ACT_LOW(PIN_VSYNC);
	int_cfg_vsync.int_bit_debounce = GPIO_INT_BIT_DIS_DEBOUNCE(PIN_VSYNC);
	clk_ptr->gpio_control(GPIO_CMD_SET_BIT_INT_CFG, (void *)&int_cfg_vsync);

	bit_isr_vsync.int_bit_ofs = PIN_VSYNC;
	bit_isr_vsync.int_bit_handler = vsyncHandler;
	clk_ptr->gpio_control(GPIO_CMD_SET_BIT_ISR, (void *)&bit_isr_vsync);

	clk_ptr->gpio_control(GPIO_CMD_ENA_BIT_INT, (void *)(1 << PIN_VSYNC));
}
```

ericwu13, 2019-05-17:
Hi, I still couldn't use interrupt pin correctly, I used your code, and change the configuration to falling edge trigger. The function just can't be triggered, but the signal is working. Do you have any idea about it? I wonder if the problem is that I didn't use the right pin to configure interrupt?

```C
int_cfg_vsync.int_bit_type = GPIO_INT_BIT_EDGE_TRIG(PIN_VSYNC);
int_cfg_vsync.int_bit_polarity = GPIO_INT_BIT_POL_FALL_EDGE(PIN_VSYNC);
```

closed 2019-07-26 by vonhust