foss-for-mips-arc-processors/embarc_osp #110: Whether ARC IoTDK can connect to webcam and apply image processing?

CWLiang, 2019-05-23:
# Issue Summary
- **Type**: *Bug* | *Enhancement* | *Question*
- **Category**: *ARC HAL* | *Device HAL* | *BSP* | *OS* | *Middleware* | *Library* | *Example* | *Build System* | *Tools* | *Documentation* | *Installation* | *Usability*
- **Priority**: *Urgent* | *High* | *Medium* | *Low*
- **Release Version**: Specified version such as 2016.05 or All

--------------------------------------------------------------------------------
# Bug

## Development Environment
- **HOST OS**
    Input the host os name and its version.
    For example, Windows 7, 64Bit | Debian Linux 8.6, 64Bit

- **TOOLCHAIN**
    Input the toolchain name and its version.
    For example, MWDT L-2016.06 | ARC GNU 2016.03.
    You can also put the compiler version here.

- **BOARD**
    Input the board name and its version.
    For example, EMSK 2.2 | AXS 103 | nSIM

- **ARC CORE**
    Input the ARC core name, such as EM7D.

## Bug Description
   Describe the bug you met in details, and tell us how to reproduce it.
   If you want to put some log information here, you can post it here or in github gist.
   If there are some screenshots, you can post outside of github, and provide links here.

--------------------------------------------------------------------------------
# Enhancement
   Describe the enhancement you suggest to add in details.
   Explain why you want this enhancement to be added, what are the pros and cons?
   Do you have any initial implementation about this enhancement, if yes, post link here.

--------------------------------------------------------------------------------
# Question
   Input your question in details here.

CWLiang, 2019-05-23:
I wonder that if ARC IoTDK can connect webcam and apply image processing?
Is there any sample project or code?

vonhust, 2019-07-26:
you cannot use ARC IoTDK to connect Webcam because no usb driver support and limitted resource and performance of ARC IoTDK

closed 2019-07-26 by vonhust