foss-for-mips-arc-processors/embarc_osp #111: Read PIN input value without using gpio_read() in Iotdk

ericwu13, 2019-05-24:
# Issue Summary
- **Type**: *Question*
- **Category**: *Device HAL*
- **Priority**: *High* 
- **Release Version**: Specified version such as 2016.05 or All
- **BOARD**
    Iotdk
--------------------------------------------------------------------------------
# Question
   Hi, I wonder if there a simple codes to read pin value without calling gpio_read function. Since if we have to call gpio_read function, the processor would need to allocate a stack for the function, do another of unnecessary things. If there is a way like AVR that accesses the PIN by the below.

I want to have a faster pin access time in IOTDK. Thanks.
```C
uint8_t input_value = PIND&4 // get the input value of pin number 2 at port D
```

IRISZZW, 2019-05-26:
when you use the `gpio_read` function, you will finally call `_lr( (unsigned)(dev->reg_base + EXT_PORTA) );//(gpio.c)`
for example，PIN GPIO_8B0_3:
`dev->reg_base = AR_IO_GPIO_8B0_SWPORTA_DR = 0x80017800;//(dfss_gpio_obj.c)`
`#define EXT_PORTA       (0x50)  /* GPIO External Port A Register *///(gpio.c)`

so, you can do it:

`uint8_t input_value = _lr(0x80017850) & (1 << 3); //pin num 3 at gpio_8b0`

before read, make sure you have done the initial for pin.

ericwu13, 2019-05-27:
Thanks. The speed is significantly increased. But instead of using _lr, which will cause linking error, we should use:
```C
uint8_t value = _arc_aux_read(0x80017850) & (1 << 3) // read pin num 3 at gpio_8b0
```

closed 2019-05-27 by ericwu13