foss-for-mips-arc-processors/embarc_osp #129: Unclear document for baremetal/bootloader

YuguoWH, 2019-08-01:
# Issue Summary
- **Type**: *Enhancement*
- **Category**: *Documentation*
- **Priority**: *High*
- **Release Version**: 2019.06-RC1

# Enhancement
   In document, the example command for iotdk is like this:

> $ gmake BOARD=iotdk BD_VER=10 CUR_CORE=arcem9d TOOLCHAIN=mw LOCATION=eflash bin

`LOCATION=eflash` has no effect to makefile, need to modify this.

ericwu13, 2019-08-17:
Do you know how to let bootloader downloaded into eflash?

YuguoWH, 2019-08-19:
@IRISZZW

IRISZZW, 2019-08-19:
> Do you know how to let bootloader downloaded into eflash?

step 1: 

generate a bootloader.bin file witch can run in eflash.

add `APPL_DEFINES += -DUSE_APPL_MEM_CONFIG` in makefile.

add 
```
#ifdef BOARD_IOTDK
#define REGION_ROM	REGION_EXT_ROM
#endif
```
in appl_mem_config.h

run command **make BOARD=iotdk bin** to generate bootloader.bin file and copy it to SD-Card in iotdk board.

step 2: 

use **flash** cmd in ntshell to write bootloader.bin into eflash.

undo the changes in makefile, run command **make BOARD=iotdk run**.

run command in ntshell **flash -eflash bootloader.bin**.

then reset the board.

closed 2019-08-20 by YuguoWH

L-leopold, 2020-08-31:
Sorry, I did not fix this problem:
The command I used:
First ,I modified the makefile and changed the USE_MCUBOOT=0, because USE_MCUBOOT=1 will prompt error
Second I add APPL_DEFINES += -DUSE_APPL_MEM_CONFIG in makefile. I did not modified the appl_mem_config.h ,because it cannot find REGION_EXT_ROM
When I did like that, it compiled good.But....
I copyed the .bin to my SDcard ,and made run successfully,and press the reset button ,and It didn't work.
This is the error:
**COM1>embARC Build Time: Aug 31 2020, 04:54:56
Compiler Version: Metaware, Clang 9.0.1
FatFS initialized successfully!
boot.json open error. use default bootloader

Press any button on board to stop auto boot in 5 s
. . . . .
Start loading 0:\boot.bin from sdcard to 0x20000000 and run...
led out: f, ff**
Could you please help me ?

L-leopold, 2020-08-31:
I can run the boadloader successfully,but when I add my own project bin in sdcard, it doesn't work. and its tip is like this:
COM1>embARC Build Time: Aug 31 2020, 04:54:56
Compiler Version: Metaware, Clang 9.0.1
FatFS initialized successfully!
boot.json open error. use default bootloader

Press any button on board to stop auto boot in 5 s
. . . . .
Start loading 0:\boot.bin from sdcard to 0x20000000 and run...
led out: f, ff
embARC Build Time: Aug 31 2020, 04:54:56
Compiler Version: Metaware, Clang 9.0.1
FatFS initialized successfully!
boot.json open error. use default bootloader

Press any button on board to stop auto boot in 5 s
. . . . .
Start loading 0:\boot.bin from sdcard to 0x20000000 and run...
led out: f, ff
Press any button doesn't work ,sorry  to bother  you.

PR #135 (2019-08-20): example: bootloader: fix unclear document
fix unclear document for bootloader example.

Fixes #129
commit 3a5b2924 (2019-08-20): example: bootloader: fix unclear document

fix unclear document for bootloader example.

Fixes #129