foss-for-mips-arc-processors/embarc_osp #134: Question:

liuy3, 2019-08-13:
# Issue Summary
- **Type**: *Question*
- **Category**:  *Device HAL* 
- **Priority**: *High*
- **Release Version**: 2019.06-RC1

--------------------------------------------------------------------------------

## Development Environment
- **HOST OS**
    Windows 10, 64Bit

- **TOOLCHAIN**
    gnu

- **BOARD**
    IoTDK
- **ARC CORE**
    
--------------------------------------------------------------------------------
# Question
   It seems that there is no Device HAL for ADC ports. I've found two files about ADC at https://github.com/foss-for-synopsys-dwc-arc-processors/embarc_osp/tree/master/device/ip/subsystem/adc. 
However, I got no clues to use it directly. Is there any way I can take advantage of an analog port to get an input signal? Otherwise, those arduino analog ports are useless at this point. 

Please reply me soon. It's getting close to the contest. Thanks.

IRISZZW, 2019-08-13:
There is no implement of subsystem adc ip in all boards now, so there is no HAL for ADC ports now.
in iotdk board there is an other adc ip, and corresponding drvier file is in [smic_adc](https://github.com/foss-for-synopsys-dwc-arc-processors/embarc_osp/tree/master/board/iotdk/drivers/smic/smic_adc)

IRISZZW, 2019-08-13:
example code for smic adc in iotdk board:
```
#include "embARC.h"
#include "embARC_debug.h"

ADC_DEFINE(adc_test, ADC_INT_NUM, ADC_CRTL_BASE, NULL);

void arduino_pin_init(void)
{
	io_arduino_config(ARDUINO_PIN_AD0, ARDUINO_ADC, IO_PINMUX_ENABLE);
	io_arduino_config(ARDUINO_PIN_AD1, ARDUINO_ADC, IO_PINMUX_ENABLE);
	io_arduino_config(ARDUINO_PIN_AD2, ARDUINO_ADC, IO_PINMUX_ENABLE);
	io_arduino_config(ARDUINO_PIN_AD3, ARDUINO_ADC, IO_PINMUX_ENABLE);
	io_arduino_config(ARDUINO_PIN_AD4, ARDUINO_ADC, IO_PINMUX_ENABLE);
	io_arduino_config(ARDUINO_PIN_AD5, ARDUINO_ADC, IO_PINMUX_ENABLE);
}

int main(void)
{
	uint16_t adc_value[6];
	arduino_pin_init();

	smic_adc_open(adc_test);

	while (1)
	{
		for(uint8_t i = 0; i < 6; i++) {
			smic_adc_read_polling(adc_test, i, &(adc_value[i]));
		}
		EMBARC_PRINTF("0: %4d, 1: %4d, 2: %4d\r\n", adc_value[0], adc_value[1], adc_value[2]);
		EMBARC_PRINTF("3: %4d, 4: %4d, 5: %4d\r\n", adc_value[3], adc_value[4], adc_value[5]);
		board_delay_ms(100,1);
		EMBARC_PRINTF("\x1b[2k\x1b\x45");// CRT control cmd.
		EMBARC_PRINTF("\x1b[2k\x1b\x45");// CRT control cmd.
	}

	return E_SYS;
}
```

liuy3, 2019-08-13:
It works for me. Much appreciated!

closed 2019-08-13 by liuy3