foss-for-mips-arc-processors/embarc_osp #138: IoTDK cannot read SD card

yuansung18, 2019-08-22:
# Issue Summary
- **Type**:  *Question*
- **Category**: *Example*
- **Priority**:  *Medium* 
- **Release Version**: All

--------------------------------------------------------------------------------
## Development Environment
- **HOST OS**
   Windows 10, 64Bit

- **TOOLCHAIN**
    ARC GNU 2019.03.

- **BOARD**
    IoTDK

- **ARC CORE**
    EM9D.

# Question
   As image shown below, how can I fix "Card did not respond to voltage select!"?

[image]

yuansung18, 2019-08-24:
And I change another SD card, it showed the message as the image.
[image]

Li-xiaofei, 2020-03-28:
I also met such problem, did you solve it later？Can your board read SD card now？Please tell me the solution. Thank you.