foss-for-mips-arc-processors/embarc_osp #146: IOTDK SDcard can  not read my own project

L-leopold, 2020-08-31:
# Issue Summary
- **Type**: *Question*
- **Category**: *ARC HAL* | *Device HAL* | *BSP* | *OS* | *Middleware* | *Library* | *Example* | *Build System* | *Tools* | *Documentation *Usability*
- **Priority**: *Urgent* | *High* |
- **Release Version**: Specified version such as 2016.05 or All

--------------------------------------------------------------------------------
# Bug

## Development Environment
- **HOST OS**
Win10
- **TOOLCHAIN**
MWDT
- **BOARD**
IOTDK
- **ARC CORE**
EM9D
## Bug Description
Sorry, I can not launch my own bin in SDcard. 
The command I used:
First ,I modified the makefile and changed the USE_MCUBOOT=0, because USE_MCUBOOT=1 will prompt error.
Second I add APPL_DEFINES += -DUSE_APPL_MEM_CONFIG in makefile. I did not modified the appl_mem_config.h ,because it cannot find REGION_EXT_ROM
When I did like that, it compiled good.But....
I copyed  bin of bootloader to my SDcard ,and made run successfully,and press the reset button .But It didn't work.
This is the error:
**COM1>embARC Build Time: Aug 31 2020, 04:54:56
Compiler Version: Metaware, Clang 9.0.1
FatFS initialized successfully!
boot.json open error. use default bootloader

Press any button on board to stop auto boot in 5 s
. . . . .
Start loading 0:\boot.bin from sdcard to 0x20000000 and run...
led out: f, ff**

I can run the boadloader successfully,but when I add my own project bin in sdcard, it doesn't work. and its tip is as stated above
Press any button doesn't work ,sorry to bother you.Could you please help me ?
--------------------------------------------------------------------------------
# Enhancement
   Describe the enhancement you suggest to add in details.
   Explain why you want this enhancement to be added, what are the pros and cons?
   Do you have any initial implementation about this enhancement, if yes, post link here.

--------------------------------------------------------------------------------
# Question
   Input your question in details here.

IRISZZW, 2020-09-01:
Can you provide the generated linker script and map files?
These file usually located in the output folder, named *.map and *.ldf.

L-leopold, 2020-09-01:
[mw_arcem9d.zip](https://github.com/foss-for-synopsys-dwc-arc-processors/embarc_osp/files/5153651/mw_arcem9d.zip)

IRISZZW, 2020-09-01:
Have you flash the bootloader into the elfash ?
https://github.com/foss-for-synopsys-dwc-arc-processors/embarc_osp/issues/129#issuecomment-522472862

L-leopold, 2020-09-01:
Yes, of course , I follow the procedure of the instruction.
Use this command : flash -eflash test.bin
and next I copyed my own bin into SDcard ,and rename it to boot.bin
next I put the reset button ,but it  doesn't respond,so I have to rerun the bootloader example.
it tips:
[image]

I think it needs *.json document. So I copyed the json document of bootloader example in Sdcard.
But when I put the json file in it  , it jumped into ntshell directly.

closed 2020-12-19 by L-leopold