foss-for-mips-arc-processors/embarc_osp #158: AGU context preserving for FreeRTOS tasks.

iguryanov, 2021-06-25:
# Issue Summary
- **Type**:*Enhancement* 
- **Category**:  *OS* 
- **Priority**:  *High* 
- **Release Version**: 2019.06

--------------------------------------------------------------------------------
Hi, 

as we discussed on a call, here is a request to add AGU/DSP task context save/restore in FreeRTOS kernel for our em9d customer. 
This would allow to use more than one MLI tasks (like face and speech recognition together). 

Regards,
Igor.

iguryanov, 2021-06-25:
Perhaps this can be done conditionally, with some extra task property indicating that it needs that extra context handling.

YuguoWH, 2021-06-28:
Hi @iguryanov , do you know if there are any technical details discussed/decided on that call? 
AGU have small/medium/large configuration, do we need to support them all? 
And by the way, which branch/version our customer is using?

iguryanov, 2021-06-30:
Hi Yuguo, 

customer has large AGU. It might have sense to make that configurable - maybe somebody else would use the feature. 
Alexey demonstrated that embARC has shared macro defined for saving/restored registers, and FreeRTOS just calls them. 

I don't might to change shared macro so we don't have to touch FreeRTOS. Might be simpler at a cost of bigger overhead where it is not needed. 

Regards,
Igor.

YuguoWH, 2021-07-01:
Hi @iguryanov , I implement a straight forward AGU save/restore feature which is usable for small/medium/large configs. I put it here on a saparate branch; https://github.com/foss-for-synopsys-dwc-arc-processors/embarc_osp/tree/feature/agu

iguryanov, 2021-07-01:
Thank you Yuguo, 

do I understand correctly that if not configured explicitly, tcf would be checked for core_config_agu_* and ARC_FEATURE_AGU* would be set? 

Looking at arc_feature_config.h it seems if user by mistake manually provide ARC_FEATURE_AGU (only) then core_config_agu_* are not checked and specific small/medium/largo macro would not be defined. 

Regards,
Igor.

YuguoWH, 2021-07-02:
Hi @iguryanov , you are right. In current implement when user provide ARC_FEATURE_AGU, small agu will be the default configuration. Otherwise the medium/large macro should be provided by user.

iguryanov, 2021-07-02:
I see. Thank you!

closed 2021-07-02 by YuguoWH