foss-for-mips-arc-processors/embarc_osp #160: How to read from sd memory?

darko262, 2021-07-11:
Hi i'm new
I am working with THE starter kit with embarc_osp
and I want to create a file, how can I create a file? How can I read from the SD card?
fopen () doesn't create the file for me.

Another question is how do you use the network cable to connect to the internet with the board?

YuguoWH, 2021-07-13:
Hi @darko262 , f_open() with FA_CREATE_NEW should be able to create a file. 
If you still have a problem, could you please give more detail?
At present embARC OSP does not support wired internet access.  There are several wireless methods instead.

darko262, 2021-07-13:
Hello, thanks for answering.
the problem that does not ask me to define the variables, what libraries does FA_CREATE_NEW use?
sorry if my questions are very naive
[image]

YuguoWH, 2021-07-14:
Hi @darko262 , are you running one of the examples or it is your own project? The library is fatfs which has already port to embarc OSP, you could find it at embarc_osp/middleware/fatfs . Please note that in order to use fatfs, you'll need to select the middleware in your makefile. You could take embarc_osp/example/baremetal/bootloader/makefile as reference, see below
https://github.com/foss-for-synopsys-dwc-arc-processors/embarc_osp/blob/master/example/baremetal/bootloader/makefile#L32
```
# middlewares
MID_SEL = common ntshell ihex fatfs parson
```

abrodkin, 2021-07-15:
@darko262 if the problem is solved for you now please let us know or close this issue yourself. If there're still questions on your side feel free to ask and we'll try to help you with them. Though if new questions are not strictly related to this one please open another issue so that we may track them properly and not miss important input of yours.

darko262, 2021-07-15:
Thank you so much!! problem solved

closed 2021-07-15 by darko262