foss-for-mips-arc-processors/embarc_osp #162: Bluetooth hm10

darko262, 2021-07-22:
--------------------------------------------------------------------------------
# Question
hi i am testing embarc_osp / example / baremetal / ble_hm1x /. I have communication from the cell phone to the device, however to write from the bluetooth to the cell phone it does not print anything on the cell phone. Why is this? Is there a function to show what I write from the device to the cell phone?

YuguoWH, 2021-07-23:
Hi, are you using BLE app on your cellphone (and which one is it)? BTW which version of embarc OSP are you using? I'd suggest 2019.06 release for now because some latest changes have not been thoroughly tested.

darko262, 2021-07-23:
thanks for the answer, the application is serial bluetooth terminal. So the example says if I write from the application it should show me the message from the cell phone on the screen, and then send the same message to the cell phone from the device, but the latter does not do it, it is supposed to write it but I do not receive anything at mobile.
using version 2019.06

YuguoWH, 2021-07-23:
Thanks for your information, I'll take a look and give you feed back ASAP.

YuguoWH, 2021-07-26:
Hi @darko262 , I was unable to reproduce the issue. Using 2019.06, testing embARC_OSP/example/baremetal/ble_hm1x example. 
HM-10 module connected to EMSK(ver23, arcem7d) PMOD J1. BLE app is LightBlue on iOS (I don't have serial bluetooth terminal on Android now). the logs which are from last part of the test are shown below. 
```
...  
                                                                                
Try to reset the baudrate to 9600bps                                            
Do send and receive test under baudrate:9600bps                                 
Start HM1X-BLE Send Receive Test, please connect to HM1X-BLE using its app.     
The main loop will receive data from HM1X, and send back what it received.      
hellohello, embARC osp testing here on 2021/07/26 using LightBlue

```

darko262, 2021-08-04:
Thanks for your information

closed 2021-08-04 by darko262