foss-for-mips-arc-processors/embarc_osp #165: update date and time

darko262, 2021-08-07:
# Question
I occupy the library <time.c> but the time and date are not updated.
How do I update the date and time on the device?

YuguoWH, 2021-08-09:
Could you please provide more details?

darko262, 2021-08-09:
I am capturing data from the digital analago converter and I want to add the date and time, for this I was occupying the library <time.c> in which it shows me the date and time but June 2020.
My question is if the date and time of the device can be updated

darko262, 2021-08-10:
also if I save a file on the card I get that date that I mentioned earlier

YuguoWH, 2021-08-10:
> I was occupying the library <time.c>

not sure what this means. 
And I'd like to know which function and which toolchain you are using, because here we could see the handling is different for MetaWare and GNU toolchain:

https://github.com/foss-for-synopsys-dwc-arc-processors/embarc_osp/blob/87e8c1dd7b5d757622bec83b6eabf3261ca7af30/library/clib/embARC_syscalls.c#L584-L626