foss-for-mips-arc-processors/embarc_osp #181: arc_excaption priority manipulation

vandit86, 2024-07-05:
``Hello, 
Can anyone explain why some getter/setter functions in arc_excaption.c, for example  int_pri_set(), int_ipm_set(), int_ipm_get()..,  change the interrupt value passed into the function by adding or subtracting INT_PRI_MIN .. ?????

example : 
````
int32_t int_ipm_get(void)
{
	return ((int32_t)arc_int_ipm_get() + INT_PRI_MIN);
}

```

abrodkin, 2024-07-05:
@vandit86 good question, we'll look into that and will get back to you with comments.

xxkent, 2024-08-27:
Unfortunately, the code lacks comments from the author about the reasons for shifting the priorities of the specified interrupts.

Judging by the code, the only reason for all this manipulation is to reserve lower priority values ​​as error values 
​​when exiting functions. For example:

```
#ifndef INT_PRI_MIN
#define INT_PRI_MIN	(-2)		/*!< the minimum interrupt priority */
#endif

#define NUM_EXC_CPU		16
#define NUM_EXC_INT		9
#define NUM_EXC_ALL		(NUM_EXC_CPU + NUM_EXC_INT)

Inline uint32_t arc_int_pri_get(const uint32_t intno)
{
	_arc_aux_write(AUX_IRQ_SELECT, intno);
	return _arc_aux_read(AUX_IRQ_PRIORITY) & 0xf;
}

int32_t int_pri_get(const uint32_t intno)
{
	if (intno >= NUM_EXC_CPU && intno < NUM_EXC_ALL) {
		return (int32_t)arc_int_pri_get(intno) + INT_PRI_MIN;
	}

	return 0;
}

int32_t secureshield_int_pri_get(uint32_t intno)
{
	if (arc_in_user_mode() == 0) {
		return int_pri_get(intno);
	} else {
		return SECURESHIELD_SECURE_CALL(SECURESHIELD_SECURE_CALL_INT_EXC, "", SECURESHIELD_INT_EXC_PRI_GET, intno);
	}
}

int main(void)
{
...
	val = secureshield_int_pri_get(INTNO_TIMER0);
	EMBARC_ASSERT(val < 0);
...
}
```

In this call sequence: main()->secureshield_int_pri_get()->int_pri_get(), value '0' means error.
Offset INT_PRI_MIN reserves abs(INT_PRI_MIN) numbers for error codes.

xxkent, 2025-02-03:
Closing this based on the previous comment.